particle-iot/electron #18: Add ESD protection to the u.FL connector area

brandoaire, 2015-08-13:
We discussed with a u-blox tech the possibility of adding ESD protection to the u.FL connector to prevent damage to the internal TX and RX lines of the u-blox module. The details are included in the application notes for the G350 and U260/270.  It would include adding a single capacitor and an inductor, I think. @technobly would know.

technobly, 2015-08-14:
[image]

C3 and L2 are the required ESD components.
[p.109 - SARA-G3 and SARA-U2 series - System Integration Manual](https://www.u-blox.com/sites/default/files/products/documents/SARA-G3-U2_SIM_(UBX-13000995).pdf)

### C3
SARA:U2: 15 pF Capacitor Ceramic C0G 0402 5% 50 V - GRM1555C1H150J - Murata
SARA-G3: 0 ohm jumper 0402

### L2
SARA:U2: 39 nH Multilayer Inductor 0402 (SRF ~1 GHz) - LQG15HN39NJ02 - Murata
SARA-G3: Do not install

### C2 
may not be necessary, double check with u-Blox.

brandoaire, 2015-08-20:
After further discussion, we've decided that that the marginal amount of protection that this will add to the PCB is not worth the expanded board size and potential impact on the RF network.  No need to implement.

closed 2015-08-20 by brandoaire