particle-iot/electron #27: Reduce size of the diode pad

brandoaire, 2015-08-28:
[image]

Line item #3 on beta unit production feedback

ericyuanyong, 2015-09-02:
I checked the footprint of this diode and find out the package on PCB is the same with the recommendation one in datasheet of this diode.So i don't think we need to change it. We also used this package in Core and Photon. So I'm not quite sure this is really a problem or not.

brandoaire, 2015-09-04:
@ericyuanyong @technobly (Brett) has some reasons to believe that a change is necessary--he's going to add more details here, but if you'd give the changes a try and commit the edits for review on your day Monday, it'd be appreciated!

technobly, 2015-09-04:
@ericyuanyong the thing that's happening with the diode that because there is no thermal relief, copper has filled in completely around the pad.  Because the solder mask is larger than the defined top copper pad, the resulting composite PCB has a much larger top copper pad that was intended.  This is allowing the part to slide around too much during reflow.

To resolve this, we can create a version of the diode that is to be used in no thermal relief situations, and add a `_NT` (no thermal) suffix to the part name.  This will be handy for other parts as well so we should standardize the suffix.  We just need to back calculate the sizes since the solder mask is auto generated by Eagle.  Reduce the new top copper pad size for the new _NT part so that the resulting new solder mask is the same size as the old top copper.  I want to say the soldermask is about a 4 mil anualar ring around the pad size (might be 0.1mm though)... but you'll have to check.

closed 2015-09-18 by brandoaire