particle-iot/electron #35: Confusing Silkscreen

brandoaire, 2015-09-21:
We've received feedback from beta testers that the silkscreen markings are confusing, since they lie in between pins.  Potential solutions:

- adding a small triangular arrow pointing up or down next to the label?

monkbroc, 2015-09-21:
Another reason for the confusion is that some labels are "above" the pin (VIN, 3V3) while most are "below".

brandoaire, 2015-09-21:
Hey @monkbroc that specific issue has been fixed in subsequent versions of the hardware.

monkbroc, 2015-09-21:
:+1: @brandoaire

technobly, 2015-09-24:
I literally spent 60 seconds making this so please forgive me!  But what about just a blob of silkscreen that helps to tie the label to the pin above it?  Arrows would be hard to see... but this might work.
[image]

monkbroc, 2015-09-24:
That would be good. I would have understood that intuitively.

kennethlimcp, 2015-09-25:
I just realized the header are labeled on the inside. Can we have them shown outside instead?

[image]

jenesaisdiq, 2015-09-25:
Obviously a manufacturing error, @kennethlimcp. Won't be like that in production.

technobly, 2015-09-25:
You've Cat 2 B Kitten Me!!! WAT.  I knew we were working on this and scratch tested samples.  I've been developing with 2 production Beta Electrons and have not noticed those labels until RIGHT MEOW.  :joy_cat:

kennethlimcp, 2015-09-25:
This might be a good resolution in addition to a silkscreen improvement :)

monkbroc, 2015-09-25:
That is bananas! I've never see that done before! I thought about labels on the side of the headers but didn't realize it could actually be done :smile: 

Yeah, consider this issue closed! (just make sure the labels are on all pins instead of just some)

brandoaire, 2015-09-25:
Awww we were hoping to keep this as a surprise :-) don't tell!

technobly, 2015-09-25:
Shred the files, shred the files!!

kennethlimcp, 2015-09-25:
:fire :rabbit: gone!

mohitbhoite, 2015-11-30:
closing this issue now, header labels FTW

closed 2015-11-30 by mohitbhoite