particle-iot/electron #37: Make u-blox GPS communication pins available

monkbroc, 2015-09-24:
The u-blox cellular module can work in tandem with a u-blox GPS module through the DDC pins (SDL and SDA, pins 26, 27). Can you consider making those pins available to the outside world through user pins? B0 and B1 seem convenient from the layout perspective.

It will make it easier to add an integrated GPS solution.

DDC pins allow -0.3 V to 3.6 V and are tristate without pull-up at reset so they wouldn't interfere with B0/B1 unless enabled through u-blox AT commands.

[image]

andyw-lala, 2015-09-24:
I see where you're going with this.. :)

technobly, 2015-09-24:
>I see where you're going with this.. :)

I see what you did there ;-)

---

This is a great suggestion as it would make it easier to operate GPS modules, the u-blox module will automatically intialize and talk to u-blox GPS receivers.  If you don't use a u-blox module though it might not work well?  There also might be some other required signals to expose and components to add to make this really a drop in solution:
[image]
cc: @mohitbhoite @jenesaisdiq

mohitbhoite, 2015-09-24:
This is a good suggestion and can be executed with minimal intrusion to the existing setup. I wonder we if have enough bandwidth to test this setup before committing a hardware change.

alan707, 2015-09-24:
> This is a good suggestion and can be executed with minimal intrusion to the existing setup. I wonder we if have enough bandwidth to test this setup before committing a hardware change.
+1 that would be awesome guys!

monkbroc, 2015-09-25:
> I don't think we'd want to rely on that fact... two pins shorted together that can be set to opposing states (HIGH/LOW), will be set to opposing states eventually :) :zap: :fire:

My point was: if somebody doesn't enable the DDC pins they will be tristate so the user firmware can use B0 and B1 as either inputs or outputs. If someone uses the DDC pins, the firmware can force B0 and B1 to be tristate to avoid the :zap: 

Regarding the other signals, could GPIO2, 3, 4 also be made available on other header pins?

All the orange components in the picture could be added on an external board as not to increase the cost of the Electron itself.

technobly, 2015-09-28:
I ran across this and thought I would share.  I just skimmed it a bit to learn more about what makes GNSS advantageous.  I didn't get a full picture, but it seems like there are some power saving features that may be duplicatable without a GNSS receiver and also sounded like there is a way that the cellular modem can automatically download some data over the cellular network instead of from satellites to speed up the initialization process.  Not sure if I skimmed that correctly though.
https://www.u-blox.com/sites/default/files/GNSS-Implementation_AppNote_(UBX-13001849).pdf